korn0055/bean-bag-toss #13: ambient light sensor

korn0055, 2015-03-18:
Implement A/D routines to read ambient level from photo resistor and adjust "off" brightness level

korn0055, 2015-03-19:
There is a lot of noise (>100mV p-p) on the photo resistor with just one emitter and the small LED strip running.  From testing indoors with a 100k series resistor, it looks like the threshold will need to be a couple hundred millivolts, so some filtering will be needed.  The plan is to use a first-order IIR filter: y[0] = (1-alpha)*x[0] + y[-1] with alpha close to unity in order to filter the noise.  A warm-up time will be required before the filter output can be used to make any decisions.

NothingButGunpowder, 2015-03-19:
You could also try stacking a cap on top of the existing resistor to filter the noise.  Is the noise synchronous with the other devices pulsing or could it be noise from overhead lights being picked up at 60 Hz?